Opendigitalradio/ODR-DabMod #9: sse with gcc 4.9.2

Hoernchen, 2015-04-12:
Building with gcc 4.9.2 fails because the sse check doesn't work, see https://bugzilla.redhat.com/show_bug.cgi?id=1092991#c1 for an explanation. Adding CFLAGS="-msse2" to configure.ac before the sse test runs fixes this for me.

mpbraendli, 2015-04-12:
I have been compiling ODR-DabMod with GCC 4.9.2 for some time now. On this machine the -msse and -msse2 flags are correctly defined by the autotools.

What distribution are you on ? (maybe I can reproduce)

using gcc (GCC) 4.9.2 20150304

mpbraendli, 2015-04-12:
Maybe it would be better to replace the ad-hoc check by http://www.gnu.org/software/autoconf-archive/ax_ext.html

I will look into this solution.

Hoernchen, 2015-04-12:
*cough* msys2/mingw32, I was curious if I'd get it to build. (The answer is yes.)

mpbraendli, 2015-04-14:
I have taken as an action to improve for the autoconf check.

I'm not sure cygwin and it's silblings is a target we want to support, but it would be nice if it would work.

closed 2015-04-14 by mpbraendli