Opendigitalradio/ODR-DabMod #13: Low SNR in I/O raw file

andimik, 2017-04-04:
I used the ini file from https://www.welle.io/devices/rawfile to create an I/O raw file from an ETI.

But I get a file with a very small SNR (screenshots taken from qt-dab and welle.io)

[image]

[image]

How can I improve it?
(Albert himself recommended to open an issue here)

rashuk, 2017-04-04:
Hello,

It probably has something to do with welle.io expecting only unsigned 8-bit
(0-255) interleaved samples, not the signed 8-bit samples generated by
odr-dabmod.

I'd imagine you're losing a lot of dynamic range as it is seems to be able
to use the positive samples (0-127) but everything else (-1 to -128) will
be out of range.

That is where your SNR has gone! :-)

Best regards,

Rash.

rashuk, 2017-04-04:
Actually, not quite...

You've only lost 1-bit of resolution because the highest integer (127) can
be represented with 7-bits, so you've only lost 6dB of theoretical dynamic
range. I can only imagine that somehow the 'bad' negative numbers become
the noise floor, leaving you with only ~6dB SNR. That's pretty close to
what we can see on your qt-dab screenshot. Somebody clever will be able to
elaborate on this no doubt!

R.

andimik, 2017-04-04:
Could you upload a screenshot how it should look like?

In the meanwhile I'll try to convert it with sox

rashuk, 2017-04-04:
Hello,

I don't have the same tools installed I'm afraid, and I don't keep IQ files
around for long!

The below is from RF, not baseband, and with a higher resolution (14-bit)
DAC. With 8-bit samples you will see no more than the the 48 dB (8-bits*6)
between the flat top and the noise floor. Expect a few dB less. Ignore the
spurious spike in the picture and the uneven noise.

If you wanted you could open it in a gnu radio flowgraph (not forgetting to
tell it the samples are signed integers and throttle them to 2048k
sampling, then pass them to an fft display.

That should work but I've never had a reason to do it.

R.

andimik, 2017-04-04:
Yeah, your hint was the solution.

If someone else wants to try: The command is

`sox -e signed-integer -r 240000 -t raw -c 2 -b 8 DabMod.iq -e unsigned-integer -t raw -c 2 -b 8 -r 240000 DabMod_new.iq
`
(sox has to be installed of course)

[image]

I will tell Albrecht to include this ;-)

closed 2017-04-04 by andimik