Opendigitalradio/ODR-DabMod #15: OSX .configure error

dmitryelj, 2017-12-08:
Hi all,

I'm trying to build the stuff:

./bootstrap.sh - ok
./configure:
checking for SOAPYSDR... no
checking for main in -luhd... yes
checking for boostlib >= 1.54.0... yes
checking for main in -lboost_system... yes
checking for main in -lboost_thread... no
configure: error: library boost_thread is missing

Boost is installed.
Probably boost_thread check should be changed to boost_thread-mt?

mpbraendli, 2017-12-10:
I did apply the same changes to ODR-DabMod as I did to ODR-DabMux. Can you also check `next` ?

dmitryelj, 2017-12-11:
Hi. I have now errors in .configure:

checking for clock_gettime in -lrt... no
configure: error: library rt is missing

The same for AudioEnc, btw.

mpbraendli, 2017-12-11:
ODR-DabMod still uses `clock_gettime` which doesn't exist on OSX. I'm currently doing a rewrite of the SDR output code, and I will replace this by `std::chrono` which is portable. I'm afraid this will take some more time. However, this function is only used in the UHD output: you can remove the line containing `clock_gettime` from the `configure.ac` file, run the bootstrap, and configure without enabling UHD, and it should hopefully compile (unless the boost linking issue is still present...)

dmitryelj, 2017-12-11:
I was going to test it exactly with UHD :) So, I'll wait for the next version.

mpbraendli, 2017-12-11:
Does it compile with the line removed and `configure --disable-output-uhd` as suggested in my previous message?

If you have SoapySDR installed, you can still drive an SDR device using the `soapy` output. SoapySDR support is detected at `./configure` time.

dmitryelj, 2017-12-11:
No, I am getting the error with librt yet:

checking for clock_gettime in -lrt... no
configure: error: library rt is missing

mpbraendli, 2017-12-11:
You need to remove the check, and bootstrap again.

dmitryelj, 2017-12-11:
I removed the check, now getting the error during make:

*src/Buffer.cpp:33:10: **fatal error: **'malloc.h' file not found*

mpbraendli, 2017-12-11:
ODR-DabMod uses `memalign` in several places, which is deprecated and apparently not existing in `malloc.h` on OSX. I think I should replace that by `posix_memalign`. Let me try this.

mpbraendli, 2017-12-11:
Can you try again? You should also not need to remove the `librt` check anymore.

dmitryelj, 2017-12-11:
Malloc error gone, now I am getting this:

*src/Buffer.cpp:90:74: **error: **no member named 'to_string' in namespace
'std'*

            throw std::runtime_error("memory allocation failed: " +
std::to_string(ret));

*                                                                    ~~~~~^*

mpbraendli, 2017-12-15:
I've added missing `<string>` includes in all the files that use `to_string()`, it should work now. Does it still fail?

dmitryelj, 2017-12-15:
Hi,

I'm getting next error now:

*src/Utils.cpp:30:10: **fatal error: **'sys/prctl.h' file not found*

Dmitrii

mpbraendli, 2017-12-18:
Seems to compile now! https://travis-ci.org/Opendigitalradio/ODR-DabMod/jobs/317962746

dmitryelj, 2017-12-18:
> jobs/317962746

True, it compiles now, cool.

Dmitrii

mpbraendli, 2017-12-19:
Yes, I'll have to add support for UHD later. I'll close this for now.

closed 2017-12-19 by mpbraendli