Opendigitalradio/ODR-DabMod #16: Add multicast support and option to choose needed interface

piratfm, 2018-01-30:
This patch provides support for VLC-like URL's, that allows to bind specified interface and (or) choose multicast group to join/connect to.

Here is examples of URL that is supported now:
udp://:12000 - bind to default interface and receive data from port 12000
udp://192.168.1.22:12000 - bind to  interface with IP:192.168.1.22 and receive data from port 12000
udp://192.168.1.22@239.100.101.22:12000 - bind to interface with IP:192.168.1.22 and join multicast group: 239.100.101.22 and receive data from port 12000
udp://@239.100.101.22:12000 - bind to default interafce (which routes to multicast)  and join multicast group: 239.100.101.22 and receive data from port 12000

[multicast.patch.gz](https://github.com/Opendigitalradio/ODR-DabMod/files/1679289/multicast.patch.gz)

P.S. Also replaced c++11 regex by simple string parser due my ubuntu-14.04 have no full regex support.

mpbraendli, 2018-01-31:
Cool, I'll have a look soon!

piratfm, 2018-01-31:
Patch added as pull request.

closed 2018-01-31 by piratfm