Opendigitalradio/ODR-DabMod #18: Using C++11 instead of boost

piratfm, 2018-01-31:
I have some small modifications, that allows to migrate from boost to C++11 this made for having possibility to compile modulator software for LEDE/OpenWRT, but it's not fully complete (especially for telnet remote control and USRP driver), maybe it will be interesting to finalize it, so embedded devices will have less dependencies.

Here is diff on how it can be done:
https://github.com/piratfm/ODR-DabMod/commit/af1450beee6e9d43e39aeb74b9c72749e37250d6

mpbraendli, 2018-01-31:
Replacing boost with std is welcome, but I'm unhappy about `std::thread` missing the `interrupt()` functionality. On my system it is actually necessary otherwise it will not stop properly. It has to be replaced by another way to wake up the sleeping thread.
I have a long flight on the 3rd, I'll look into it then. Of course, feel free to push more commits to your `boost_to_cpp11` branch, I will follow it (No need to open a new pull request).
Thanks!

piratfm, 2018-01-31:
Since we already having dependency for posix threads, i'm thinking we can use it for creating threads that must have possibility to cancel like pthread_create() and after that pthtead_cancel(), so exception like in boost will be catched. Just not sure if all of threads must be cancelable.

mpbraendli, 2018-01-31:
Not all need to be cancelled. I've already converted most to std, except those that used interrupt().

I guess we can call pthread_cancel on the std::thread native handle, boost is probably doing the same. I wonder however how the cancellation itself gets done, maybe we need to do something special inside the thread function.

I'm OK with assuming pthread, if one day we have to support another thread implementation we'll take care of it then.

PS: License.txt from the INI lib needs to be added. Could you put into lib/ instead of /src to distinguish between our code and 3rd party code?

mpbraendli, 2018-02-01:
branch `boost_to_cpp11` is now in this repository too, and you have push rights. We can complete this work here.

mpbraendli, 2018-02-01:
My assumption that boost uses pthread_cancel was wrong, I checked the boost sources. I am currently wondering "how could it have worked?", and I will have to look into it further.

mpbraendli, 2018-04-16:
Review of goals: instead of removing boost completely, make boost optional when building ODR-DabMod for the EasyDABv3, which has part of the processing done in FPGA.
This will happen in `next` and the branch `boost_to_cpp11` will not be used anymore.

mpbraendli, 2018-04-17:
Branch `easydabv3` now compiles without boost.

closed 2018-04-17 by mpbraendli

mpbraendli, 2018-08-06:
ODR-DabMod `next` does not require boost anymore, and the telnet server should now also work for the easydabv3 build, although I didn't try.