Opendigitalradio/ODR-DabMod #20: Ignore input errors

lars18th, 2018-02-26:
Hi,

I'm using the ODR-DabMod with the "eti-tools" package to "modulate" (IQ samples to a fifo file) from a satellite encapsultated ETI-NI bitstream.

However, time to time, I see this message in the output of the ODR-DabMod:

```
WARN:  TS Discontinuity
ERROR: Deinterleave: Bad block[1]:[2.7]: f5 b.6=0
```
or
```
ERROR: Deinterleave: Bad block[1]:[2.4]: 96 b.6=1
```

And then the program exits.

It will be possible to execute this tool **ignoring** some input errors?

piratfm, 2018-02-26:
This is na2ni/ts2na related error (bad signal).

closed 2018-02-26 by piratfm

lars18th, 2018-02-26:
> This is na2ni/ts2na related error (bad signal).

OK. But I _really need_ to continue running it. Any idea?
I don't need to "solve" the problem, I only need to continue running it... perhaps inserting some silence and doing a resynch/restart.

piratfm, 2018-02-26:
You can use ZeroMQ client as input protocol for ODR-DabMod to make reconnections from modulator to ETI stream source. As we discussed on eti-tools thread, the ZeroMQ output feature can be added to ni2na, so it will have similar output as ODR-DabMUX.

Because if stdin/stdout pipeline got broken, the app must exit, due input data will never come to modulator anymore.

lars18th, 2018-02-27:
Hi @piratfm ,

> You can use ZeroMQ client as input protocol

Great idea! However, I still need to use the PIPE connection (until ZeroMQ output feature is added to ni2na).

> Because if stdin/stdout pipeline got broken, the app must exit, due input data will never come to modulator anymore.

So, the exit of the ODR-DabMod is because the input PIPE is closed? Or because it has detected an error?

piratfm, 2018-02-27:
Yes, because it can't read from input file anymore.

lars18th, 2018-02-27:
> Yes, because it can't read from input file anymore.

So, I'll check it using a FIFO instead of a PIPE.
Thank you for the info!

lars18th, 2018-02-27:
Hi @piratfm ,

Thank you for pointing in the right direction! :smile: 
I found the problem and I created a PR for "na2ni": https://github.com/piratfm/eti-tools/pull/18
Now it works!