Opendigitalradio/ODR-DabMod #22: Commandline for filelog

lars18th, 2018-02-27:
Hi,

I'm using the ODR-DabMod with _only_ command line (that is, without the configuration file). And most of the default values are fine. However, when using this tool in a PIPE chain the STDERR is mixed with the ouput of all the running tools. So, as the configuration file has the option to enable the `filelog` and setting the `filename` I suggest to support it also in the command line.

This has sense?

piratfm, 2018-02-27:
As for me: NO

lars18th, 2018-02-27:
> As for me: NO

The problem is: It's not possible to *mix* command line with configuration file. So, or one or another; not both. So, using the command line it's impossible to log.

closed 2018-02-27 by lars18th

mpbraendli, 2018-02-28:
There's currently no way to stop ODR-DabMod from printing to stderr, even when using the configuration file. So it's not just about adding an option. I'm slightly against implementing this, because I want to encourage users to work with the more flexible configuration files.

andimik, 2020-04-06:
Remark, in some documentation there is still parameter `-C` for the configuration file mentioned. Should I send you a PR to fix the docu?

mpbraendli, 2020-04-06:
Yes, please let me know if there is an oversight. Either here as a message, or a pull request if applicable.

andimik, 2020-04-06:
http://opendigitalradio.github.io/mmbtools-doc/mmbtools.pdf

Chapter 5.1.2 suggests that an eti-file can easily be loaded on the command line and the text says, there are (still) two ways of usage ...

And http://wiki.opendigitalradio.org/ODR-DabMod#Command_line_arguments is outdated as well.

andimik, 2020-04-06:
And the help itself is outdated.

```
$ odr-dabmod --help
ODR-DabMod version v2.4.1, compiled at Apr  6 2020, 19:46:27
Compiled with features: zeromq output_soapysdr SSE 
odr-dabmod: invalid option -- '-'
Usage with configuration file:
	odr-dabmod config_file.ini

Usage with command line options:
	odr-dabmod input (-f filename -F format | -u uhddevice -F frequency) [-o offset]
	 [-G txgain] [-T filter_taps_file] [-a gain] [-c clockrate]
	 [-g gainMode] [-m dabMode] [-r samplingRate] [-l] [-h]
Where:
input:         ETI input filename (default: stdin), or
                  tcp://source:port for ETI-over-TCP input, or
                  zmq+tcp://source:port for ZMQ input.
                  udp://:port for EDI input.

```

mpbraendli, 2020-04-07:
Well, it's still true that both the command line options and the config file still exist. I don't see an immediate need to remove the options.

mpbraendli, 2020-04-07:
But the wiki is indeed outdated, I'll do something about that.

andimik, 2020-04-07:
I am sorry, you are right. I guess I used the wrong syntax and did not specify enough parameters.