Opendigitalradio/ODR-DabMod #24: Poor signal with activated TII

andimik, 2018-05-06:
I tried to activate TII but then my Sticks and radios either don't lock at all or get a very poor signal (Microspot 4 squares instead of 16).

Seems the null symbol causes a gain issue on the receiver side.

andimik, 2018-05-16:
The latest from the next. I've posted the version in the (closed) issue about edi.

andimik, 2018-05-16:
ODR-DabMod version v2.0.0-26-g9ef3824, compiled at May 13 2018

andimik, 2018-05-16:
Digital gain is 0.8

mpbraendli, 2018-05-16:
Ok. I also used `next` to do a test, enabling/disabling TII with the remote-control. I wasn't able to see any effect on my receivers :-/
Can you post your .ini file in a github gist or on https://paste.ubuntu.com/ and send me a link?

andimik, 2018-05-16:
Well, now I remember. When I started the issue I still had Ubuntu 16.04. Some days ago I re-installed a clean 18.04 Kubuntu. And tested it again right now. Seems we can close now.

closed 2018-05-16 by andimik

rashuk, 2018-05-16:
> btw: My ini is:
>
> [input]
> transport=file
> source=/dev/stdin
> loop=1
>
> [modulator]
> gainmode =1
> rate=2048000
> digital_gain=0.8
>
> [output]
> output=file
>
> [fileoutput]
> format=s8
> filename=/dev/stdout
>
> [tii]
> enable=1
> comb=4
> pattern=11
> old_variant=0
>
> more foo.eti | odr-dabmod foo.ini | hackrf_transfer -t - -f 215072000 -x
> 25 -a 1 -s 2048000 -b 1750000

Hello,

I'm surprised that this can work because you are using the native sampling
rate which does not push the image frequencies far enough into the
stop-band of the 1.75 MHz monolithic filter inside the transceiver chip.
Trust me - it looked truly awful on a spectrum analyser!.;-D

If you have sufficient cpu resources, try to operate odr-dabmod and
hackrf_transfer at a higher sampling rate, ideally at least 4.096M. It will
clean up your signal a lot and the image won't be there to clobber
reception. Disable the FIR filter if you have too because it would still
much better than the close-in image interference.

Hope this is useful!

R.

andimik, 2018-05-16:
Ok thanks, I'll try that instead.

mpbraendli, 2018-05-16:
If you read from a file, better specify the filename in the `.ini` over using `more`. `cat` would be more appropriate (pun not intended) because `more` is a pager, it could actually mess with file contents.
Thanks for reporting issues, even if this was (luckily) a false positive!

andimik, 2018-05-16:
Ok, thanks.

I also had to change `rate=2048000` in the ini-file to 4096k.

andimik, 2018-05-17:
Well, I don't see a difference between 4M and 2M beside of higher traffic and CPU usage. Signal is the same. Both work here.

mpbraendli, 2018-05-17:
How do you measure your signal? Do you have a spectrum analyser?

andimik, 2018-05-17:
Qt-dab

mpbraendli, 2018-05-17:
qt-dab doesn't show a wide enough bandwidth, so you won't see the effect Rash is talking about.

andimik, 2018-05-17:
4096k and CPU usage 40%
[image]
2048k and CPU usage 35%
[image]

For testing (and helping FMSCAN to be updated with CSV files from Andreas Gsinn's Player) this is absolutely ok.