Opendigitalradio/ODR-DabMod #25: No ctrl+c possible at EDI input

andimik, 2018-05-15:
When I use EDI as input (but I'm still trying to get a signal from Munich and Stuttgart, see https://github.com/piratfm/eti-tools/issues/12 ) then ctrl+c for stopping the ODR-DabMod does not work

```
$ odr-dabmod odr_edi_input.ini 
ODR-DabMod version v2.0.0-26-g9ef3824, compiled at May 13 2018, 22:00:00
Compiled with features: EDI SSE 
Input
  Type: edi
  Source: udp://@239.16.242.13:60013
Output
  Name: /dev/stdout
  Sampling rate: 2.0480 MHz
      Configuration parsed. Starting up version v2.0.0-26-g9ef3824
ERROR Could not set priority for modulator:1
      Setting up timestamp decoder with 0 offset
      Opening EDI :udp://@239.16.242.13:60013
      EDI input: host:0.0.0.0, source:239.16.242.13, port:60013
^C^C^C^C^C
```

But it works with fileinput, on the other hand.

mpbraendli, 2018-05-16:
Fixed in 0226ed3 (branch `next`)
Thanks for reporting the issue!

closed 2018-05-16 by mpbraendli

andimik, 2020-04-05:
@mpbraendli 

I have the same problem when the input is zmq+tcp

```
$ odr-dabmod etizmq2raw.ini > /tmp/test_zmq_tcp.raw
ODR-DabMod version v2.4.0-3-g69525b1, compiled at Mar 24 2020, 11:04:58
Compiled with features: zeromq output_soapysdr SSE 
Input
  Type: zeromq
  Source: zmq+tcp://localhost:9100
Output
  Name: /dev/stdout
  Sampling rate: 2.0480 MHz
      Configuration parsed. Starting up version v2.4.0-3-g69525b1
ERROR Could not set priority for modulator:1
      Setting up timestamp decoder with 0 offset
      Input ZeroMQ: Receiving from tcp://localhost:9100
WARN  ZeroMQ buffer low: 4 elements !
^C^C^C^C^C^C^C

```
```
$ more etizmq2raw.ini 
[input]
transport=zeromq
source=zmq+tcp://localhost:9100
max_frames_queued=100

[modulator]
gainmode =1
rate=2048000

[output]
output=file

[fileoutput]
format=u8
filename=/dev/stdout
```
I have to wait for a server timeout in order to proceed.

mpbraendli, 2020-04-06:
Thanks, there was indeed an issue. I think I'll release a new version because v2.4.0 is probably also impacted.

andimik, 2020-04-06:
Yes, confirmed. Now it closes with Ctrl+C.
Danke!

andimik, 2020-11-21:
@mpbraendli 

With v2.4.1-13-ga2d708d I again have no chance to use Ctrl+C at EDI input.

But I have not tried previous commits.