Opendigitalradio/ODR-DabMod #28: adalm pluto support

eljuligallego, 2018-05-23:
Hello,
It would incredible to have support for this board due it seem suitable for DAB and it has an incredible price(99$).
any ideas how to accomplish this?.

mpbraendli, 2018-05-24:
Ideally it would be supported through SoapySDR. There were some discussions already, but I don't know if there was a follow-up. https://github.com/pothosware/SoapySDR/issues/136

If you can drive the PlutoSDR using GNURadio (maybe with gr-iio), then you can use the ODR-DabMod ZMQ output to connect to a GNURadio ZMQ source.

If you manage to transmit using the PlutoSDR, let me know, we can post a news entry on www.opendigitalradio.org !

closed 2018-05-24 by mpbraendli

maceyldn, 2020-11-30:
Hi All,

I've tried this, I have an image built for pluto although unfortunatly, it doesn't look like the pluto has enough processing power to run odr-dabmod natively. :(

rashuk, 2020-12-01:
Hi,

Did you try compiling odr-dabmod with fast math enabled and with
optimisation flag at level 3?

The other possibility would be to port some heavy bits to the zync FPGA,
but I dare say that nobody that could do it will will want to do it that
badly!

The Zync inside the Pluto is actually dual core. If you haven't enabled
that in the bootloader already, it might help you win.

A last thought is the fir filter is costly. You can disable it, but the
close-in spectrum performance would suffer.

R

F5OEO, 2020-12-01:
Ok, after helping on receive part...a bit of challenge to run it on pluto : will try soon after (still) improving dvbs2 tv modulation.

maceyldn, 2020-12-01:
Sorry looks like I may have accidentally deleted my message..

Hi All,
I've tried this, I have an image built for pluto although unfortunatly, it
doesn't look like the pluto has enough processing power to run odr-dabmod
natively. :(

Rash, fast math is on, not tried the compiler optimisation flags. Pulled zmq out aswell as it was getting a bit big. Can try optimisation later. I can see a very broken carrier on the Spectrum Analyser! CPU was up at 90% and yup had the 2nd core enabled!

Will try without the filter too...

Watch this space!

maceyldn, 2020-12-01:
Compiled with -O3 - No real difference.
Removed the fir filter - Carrier Breakup less noticeable but still not "solid"

Any other suggestions? Would love to get this working! :)