Opendigitalradio/ODR-DabMod #30: Raspberry 3 configure error - hint

andimik, 2018-07-29:
I found out that the boost version or path (?) produces an error under a RPi3.

```
pi@raspberrypi:~/apps/ODR-DabMod $ ./configure --disable-zeromq --disable-output-uhd --disable-native
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking target system type... armv7l-unknown-linux-gnueabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for style of include used by make... GNU
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether g++ supports C++11 features by default... yes
checking whether C++ compiler accepts -Wduplicated-cond... yes
checking whether C++ compiler accepts -Wduplicated-branches... no
checking whether C++ compiler accepts -Wlogical-op... yes
checking whether C++ compiler accepts -Wrestrict... no
checking whether C++ compiler accepts -Wshadow... yes
checking whether C++ compiler accepts -Wdouble-promotion... yes
checking whether C++ compiler accepts "-Wformat=2"... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for FFTW... yes
Checking zeromq
checking if compiler needs -Werror to reject unknown flags... no
checking for the pthreads library -lpthreads... no
checking whether pthreads work without any flags... no
checking whether pthreads work with -Kthread... no
checking whether pthreads work with -kthread... no
checking for the pthreads library -llthread... no
checking whether pthreads work with -pthread... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking if more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking for SOAPYSDR... no
checking for boostlib >= 1.54.0 (105400)... yes
checking whether the Boost::System library is available... yes
configure: error: Could not find a version of the library!
pi@raspberrypi:~/apps/ODR-DabMod $ 
```

Then I found the solution 

```
./configure --with-boost-libdir=/usr/lib/arm-linux-gnueabihf
```

worked then

mpbraendli, 2018-07-30:
Yes, I don't know why this is necessary. If you find a way to improve the boost configure check please do a pull request.

Even a hint in the README would be useful.

closed 2018-07-30 by mpbraendli

andimik, 2018-07-30:
Of course, I will change the INSTALL file and make a PR to the next branch

mpbraendli, 2018-08-06:
I'll replace boost by `asio.hpp` now

mpbraendli, 2018-08-06:
The boost requirement is gone in `next`.