Opendigitalradio/ODR-DabMod #32: Output file s16 seems corrupt

F5OEO, 2018-12-06:
Maybe a misunderstanding, but output 16bit file(should be litlle endian) seems to be without negative value.

Here is the result of samples : 
First 2 lines : I/Q signed int 16bits (s16 format)
Next 2 lines : I/Q signed 8 (s8 format)

Note : running on Raspberry

[
[image]
](url)

rashuk, 2018-12-06:
Hi Evariste! ;)

Is mentioning that the other output (other than s8) are 32-bit complex
floats useful?

R.

F5OEO, 2018-12-06:
Thanks Rash, I better understand. On LimeSDR, to have less USB bandwidth, it should be better to have s16 iq instead of real. I will translate it on dabmod side or Limesdrsend.

mpbraendli, 2018-12-07:
Bonjour Evariste,

That looks wrong indeed. I guess (from your screenshot) that you opened it in audacity, so I tried the same (Raw import, signed 16-bits, little-endian, stereo, 2048Hz)
[image]

Length seems correct too.

Do you think the format converter does something wrong on 32-bit systems or on ARM?

https://github.com/Opendigitalradio/ODR-DabMod/blob/master/src/FormatConverter.cpp#L55

PS:
The SoapySDR API also presents options for over-the-wire format, maybe that's an easy way to reduce bandwidth.

F5OEO, 2018-12-07:
Replace https://github.com/Opendigitalradio/ODR-DabMod/blob/5bb71419937a4d8919ef33df088f73e6c705b028/src/FormatConverter.cpp#L57
by
  int16_t* out = reinterpret_cast<int16_t*>(dataOut->getData());

Works OK now !

mpbraendli, 2018-12-07:
[image]

But why did it work here?!?

Anyway, fixed in 5d825e7

closed 2018-12-07 by mpbraendli

rashuk, 2018-12-07:
This explains a few things...and a few very long, non-productive days...

Thanks! :-D