Opendigitalradio/ODR-DabMod #35: Modulator reverts to idle

ashelford, 2019-04-13:
Our multiplexer is not on the same site as our modulator. Occasionally, if there is a glitch in the connection the modulator reverts to idle and has to be manually rebooted to reconnect the stream. Is there are workaround for this? Is it possible, for example, to specify two inputs on the modulator so that if one stream fails it switches to the other? Alternatively, is there a way we can make the modulator automatically reconnect if idle for say a predetermined period of time (say 5 seconds)? Many thanks, Ash

mpbraendli, 2019-04-15:
Hi Ash,
how do you connect both? ZMQ?

ashelford, 2019-04-15:
Yes, ZeroMQ.

mpbraendli, 2019-04-16:
It seems that some errors on the network can lead to a situation where the receive end doesn't notice an error, as discussed here. https://github.com/zeromq/libzmq/issues/1199#issuecomment-57373160
I'll see if there's an easy workaround.

mpbraendli, 2019-04-16:
I've added an auto-timeout to the zeromq input. Can you try the latest "next" commit? That would be version v2.3.0-8-g01cb197

With the log, you will be able to see how many times the modulator restarted. Let me know if that solves your issue.

closed 2019-08-12 by mpbraendli