Opendigitalradio/ODR-DabMod #37: can't compile in cygwin

jpuigs, 2019-09-09:
Is there any way to make it work on cygwin ?

<pre><font color="#4E9A06">Jordi@PC-Jordi </font><font color="#C4A000">~/Descargas/ODR-DabMod</font>
$ ./configure --disable-zeromq --disable-output-uhd
checking build system type... x86_64-pc-cygwin
checking host system type... x86_64-pc-cygwin
checking target system type... x86_64-pc-cygwin
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether g++ supports C++11 features by default... yes
checking whether C++ compiler accepts -Wduplicated-cond... yes
checking whether C++ compiler accepts -Wduplicated-branches... yes
checking whether C++ compiler accepts -Wlogical-op... yes
checking whether C++ compiler accepts -Wrestrict... yes
checking whether C++ compiler accepts -Wno-pragmas... yes
checking whether C++ compiler accepts -Wdouble-promotion... yes
checking whether C++ compiler accepts &quot;-Wformat=2&quot;... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for FFTW... yes
Checking zeromq
checking if compiler needs -Werror to reject unknown flags... no
checking for the pthreads library -lpthreads... no
checking whether pthreads work without any flags... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking if more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... no
checking for SOAPYSDR... no
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking limits.h usability... yes
checking limits.h presence... yes
checking for limits.h... yes
checking for memory.h... (cached) yes
checking netinet/in.h usability... yes
checking netinet/in.h presence... yes
checking for netinet/in.h... yes
checking for stdint.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking sys/time.h usability... yes
checking sys/time.h presence... yes
checking for sys/time.h... yes
checking sys/timeb.h usability... yes
checking sys/timeb.h presence... yes
checking for sys/timeb.h... yes
checking for unistd.h... (cached) yes
checking for M_PIl existence... no
checking for prctl and PR_SET_NAME... no
checking for MSG_NOSIGNAL... yes
checking for SO_NOSIGPIPE... no
checking if we can add -march=native to CFLAGS... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depfiles commands

***********************************************

  Features
  Enabled: 
  Disabled:  edi prof trace output_uhd zeromq soapysdr easydabv3

  Options
  Active:  supports_march_native
  Disabled:  enable_fast_math

***********************************************

<font color="#4E9A06">Jordi@PC-Jordi </font><font color="#C4A000">~/Descargas/ODR-DabMod</font>
$ make
make  all-am
make[1]: se entra en el directorio &apos;/home/Jordi/Descargas/ODR-DabMod&apos;
  CXX      src/odr_dabmod-DabMod.o
<b>src/DabMod.cpp:</b> En la función ‘<b>int launch_modulator(int, char**)</b>’:
<b>src/DabMod.cpp:260:22:</b> <font color="#CC0000"><b>error: </b></font>el agregado ‘launch_modulator(int, char**)::sigact<b>ion sa</b>’ tiene un tipo incompleto y no se puede definir
     struct sigaction <font color="#CC0000"><b>sa</b></font>;
                      <font color="#CC0000"><b>^~</b></font>
<b>src/DabMod.cpp:261:43:</b> <font color="#CC0000"><b>error: </b></font>aplicación no válida de ‘<b>sizeof</b>’ a un tipo incompleto ‘<b>launch_modulator(int, char**)::sigaction</b>’
     memset(&amp;sa, 0, sizeof(struct sigaction<font color="#CC0000"><b>)</b></font>);
                                           <font color="#CC0000"><b>^</b></font>
<b>src/DabMod.cpp:264:36:</b> <font color="#CC0000"><b>error: </b></font>uso no válido del tipo incompleto ‘<b>struct launch_modulator(int, char**)::sigaction</b>’
     if (sigaction(SIGINT, &amp;sa, NULL<font color="#CC0000"><b>)</b></font> == -1) {
                                    <font color="#CC0000"><b>^</b></font>
<b>src/DabMod.cpp:260:12:</b> <font color="#06989A"><b>nota: </b></font>forward declaration of ‘<b>struct launch_modulator(int, char**)::sigaction</b>’
     struct <font color="#06989A"><b>sigaction</b></font> sa;
            <font color="#06989A"><b>^~~~~~~~~</b></font>
<b>src/DabMod.cpp:</b> En la función ‘<b>int main(int, char**)</b>’:
<b>src/DabMod.cpp:626:5:</b> <font color="#CC0000"><b>error: </b></font>‘<b>setenv</b>’ no se declaró en este ámbito
     <font color="#CC0000"><b>setenv</b></font>(&quot;TZ&quot;, &quot;&quot;, 1);
     <font color="#CC0000"><b>^~~~~~</b></font>
<b>src/DabMod.cpp:626:5:</b> <font color="#06989A"><b>nota: </b></font>suggested alternative: ‘<b>getenv</b>’
     <font color="#06989A"><b>setenv</b></font>(&quot;TZ&quot;, &quot;&quot;, 1);
     <font color="#06989A"><b>^~~~~~</b></font>
     <font color="#4E9A06">getenv</font>
<b>src/DabMod.cpp:627:5:</b> <font color="#CC0000"><b>error: </b></font>‘<b>tzset</b>’ no se declaró en este ámbito
     <font color="#CC0000"><b>tzset</b></font>();
     <font color="#CC0000"><b>^~~~~</b></font>
<b>src/DabMod.cpp:627:5:</b> <font color="#06989A"><b>nota: </b></font>suggested alternative: ‘<b>_tzset_r</b>’
     <font color="#06989A"><b>tzset</b></font>();
     <font color="#06989A"><b>^~~~~</b></font>
     <font color="#4E9A06">_tzset_r</font>
make[1]: *** [Makefile:1024: src/odr_dabmod-DabMod.o] Error 1
make[1]: se sale del directorio &apos;/home/Jordi/Descargas/ODR-DabMod&apos;
make: *** [Makefile:611: all] Error 2

<font color="#4E9A06">Jordi@PC-Jordi </font><font color="#C4A000">~/Descargas/ODR-DabMod</font></pre>

mpbraendli, 2019-09-13:
Unless you can find a developer who is willing to maintain ODR-DabMod running on cygwin, I'm afraid it's not possible.

jpuigs, 2019-09-15:
ok

closed 2019-09-15 by jpuigs

LarsHaack, 2020-09-09:
I also gave building for Windows 10 using cygwin a try and was successfull in the end.
All compilation issues except one were gone after I added compiler flag -std=gnu++11
I configured the build using following command
`./configure CFLAGS="-O3 -DNDEBUG" CXXFLAGS="-O3 -DNDEBUG -std=gnu++11" --enable-fast-math --disable-native --disable-output-uhd`
The remainig issue was a missing type ip_mreqn in UDPSocket::joinGroup() in file lib/Socket.cpp.
I replaced the body of the method by throw runtime_error() to see if it will be called, because I do not know in which use cases this method will be triggered.
I have not yet tried to include device output to the build. 
What I have tried so far is file input to file output and file input to ZMQ output without issues.

andimik, 2021-01-04:
> I also gave building for Windows 10 using cygwin a try and was successfull in the end.

Could you share the file?

andimik, 2021-01-05:
Because I got the same error you mentioned

```
  CXX      lib/odr_dabmod-Socket.o
lib/Socket.cpp: In member function ‘void Socket::UDPSocket::joinGroup(const char*, const char*)’:
lib/Socket.cpp:237:5: error: ‘ip_mreqn’ was not declared in this scope; did you mean ‘ip_mreq’?
  237 |     ip_mreqn group;
      |     ^~~~~~~~
      |     ip_mreq
lib/Socket.cpp:238:10: error: ‘group’ was not declared in this scope
  238 |     if ((group.imr_multiaddr.s_addr = inet_addr(groupname)) == INADDR_NONE) {
      |          ^~~~~
In file included from /usr/include/netinet/in.h:12,
                 from lib/Socket.h:40,
                 from lib/Socket.cpp:25:
lib/Socket.cpp:241:29: error: ‘group’ was not declared in this scope
  241 |     if (!IN_MULTICAST(ntohl(group.imr_multiaddr.s_addr))) {
      |                             ^~~~~
lib/Socket.cpp:246:9: error: ‘group’ was not declared in this scope
  246 |         group.imr_address.s_addr = inet_addr(if_addr);
      |         ^~~~~
lib/Socket.cpp:249:9: error: ‘group’ was not declared in this scope
  249 |         group.imr_address.s_addr = htons(INADDR_ANY);
      |         ^~~~~
lib/Socket.cpp:251:5: error: ‘group’ was not declared in this scope
  251 |     group.imr_ifindex = 0;
      |     ^~~~~
make[1]: *** [Makefile:1477: lib/odr_dabmod-Socket.o] Error 1
make[1]: Leaving directory '/home/andreas/apps/ODR-DabMod-next'
make: *** [Makefile:623: all] Error 2
```

> The remainig issue was a missing type ip_mreqn in UDPSocket::joinGroup() in file lib/Socket.cpp.
> I replaced the body of the method by throw runtime_error()

@LarsHaack 
How did you solve that?

My configure was

`./configure CFLAGS="-O3 -DNDEBUG" CXXFLAGS="-O3 -DNDEBUG -std=gnu++11" --enable-fast-math --disable-native --disable-output-uhd --disable-zeromq --disable-dependency-tracking`

LarsHaack, 2021-01-05:
I fixed the compilation issue by replacing the function body with 
`throw runtime_error(string("UDPSocket::joinGroup()  NOT SUPPORTED IN CYGWIN BUILD"));`
Just to check if this is triggered in my use case. I have not tracked down the usage of this function on odr-dabmod really.

I made a fork of odr-dabmod to keep track of my CYGWIN build changes.
https://github.com/LarsHaack/ODR-DabMod

I also committed the binary I compiled:
https://github.com/LarsHaack/ODR-DabMod/releases/tag/v2.4.1_cygwin01

The configuration used for this build was:
`./configure CFLAGS="-O3 -DNDEBUG -DCYGWIN_BUILD" CXXFLAGS="-O3 -DNDEBUG -DCYGWIN_BUILD -std=gnu++11" --enable-fast-math --disable-native --disable-output-uhd`

andimik, 2021-01-05:
Success! 

Now it works in cygwin.

Although I am NOT a programmer, I found the bug. 

See https://github.com/Opendigitalradio/ODR-DabMod/pull/42

LarsHaack, 2021-01-05:
My first attempt to fix the issue was also to simply define the missing struct.
But I was not sure if the following call to 
`if (setsockopt(m_sock, IPPROTO_IP, IP_ADD_MEMBERSHIP, &group, sizeof(group))`
would handle the additional parameters correctly.

mpbraendli, 2021-01-06:
@LarsHaack if you feel you have a clean solution that compiles under cygwin, even with missing support for multicast input, I'd be happy to merge it into ODR-DabMod, as long as it doesn't break anything. Let me know!