Opendigitalradio/ODR-DabMod #41: configure: error: Could not link against boost_thread !

rkristijanto, 2020-06-09:
Hello, 

Can you please help me configuring and compiling ODR-DabMod, I am getting **configure: error: Could not link against boost_thread !** When I configure it.

```
checking whether the Boost::Thread library is available... yes
checking for exit in -lboost_thread... no
checking for exit in -lboost_thread... (cached) no
checking for exit in -lboost_thread... (cached) no
configure: error: Could not link against boost_thread !

```

Thank you 

Ronald

```
ronald@ronald-laptop:~/Documents/Mentahan/ODR/ODR-DabMod$ ./configure
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether g++ supports C++11 features by default... no
checking whether g++ supports C++11 features with -std=c++11... yes
checking whether C++ compiler accepts -Wduplicated-cond... no
checking whether C++ compiler accepts -Wduplicated-branches... no
checking whether C++ compiler accepts -Wlogical-op... yes
checking whether C++ compiler accepts -Wrestrict... no
checking whether C++ compiler accepts -Wno-pragmas... yes
checking whether C++ compiler accepts -Wdouble-promotion... yes
checking whether C++ compiler accepts "-Wformat=2"... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for FFTW... yes
Checking zeromq
checking for zmq_init in -lzmq... yes
checking if compiler needs -Werror to reject unknown flags... no
checking for the pthreads library -lpthreads... no
checking whether pthreads work without any flags... no
checking whether pthreads work with -Kthread... no
checking whether pthreads work with -kthread... no
checking for the pthreads library -llthread... no
checking whether pthreads work with -pthread... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking if more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking for SOAPYSDR... yes
checking for UHD... yes
checking for boostlib >= 1.55.0 (105500)... yes
checking whether the Boost::Thread library is available... yes
checking for exit in -lboost_thread... no
checking for exit in -lboost_thread... (cached) no
checking for exit in -lboost_thread... (cached) no
configure: error: Could not link against boost_thread !
```

mpbraendli, 2020-06-09:
Hello,
do you have the package `libboost-all-dev` installed on your system? If not, please install and try again.
If it still doesn't work, please post the `config.log` (please use gist.github.com or pastebin)

rkristijanto, 2020-06-09:
Yes, already installed.

```
ronald@ronald-laptop:~/Documents/Mentahan/ODR/ODR-DabMod$ sudo apt install libboost-all-dev
[sudo] password for ronald: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libboost-all-dev is already the newest version (1.71.0.0ubuntu2).
The following packages were automatically installed and are no longer required:
  libzita-alsa-pcmi0 libzita-resampler1
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 26 not to upgrade.
```
And here is the **config.log**

[config.log](https://github.com/Opendigitalradio/ODR-DabMod/files/4751186/config.log)

Thank you for your help

Ronald

mpbraendli, 2020-06-09:
There seems to be something broken with your build environment, there are strange errors around line 614. SoapySDR might be installed twice, and I don't know what "linuxbrew" is and why apparently it overrides the linker.
Sorry, this is probably not an issue in ODR-DabMod itself.

closed 2020-06-09 by mpbraendli

rkristijanto, 2020-06-09:
Ok, thank's anyway. I will try to build it using a fresh instance of Ubuntu