Opendigitalradio/ODR-DabMod #43: [Cygwin only]: Configuration error: Output not specified

andimik, 2021-01-05:
Although I don't expect to get it fixed by Matthias, I'll report this issue anyway. Maybe another person has an idea what goes wrong here.

Under **Cygwin** the syntax `$ odr-dabmod.exe foo.eti -f /dev/null` is not working. Of course, I tried other names or paths or other parameters (like format, etc.)

I only get

```
Modulator runtime error:  Configuration error: Output not specified
Configuration parsed. Starting up version 2.4.1
```

I also tried `cat foo.eti | odr-dabmod.exe /dev/stdin -F /dev/stdout` without any luck.

The same syntax in **Linux** is ok.

```
$ odr-dabmod foo.eti -f /dev/null
ODR-DabMod version v2.4.1-13-ga2d708d, compiled at Nov 30 2020, 20:41:54
Compiled with features: zeromq output_soapysdr SSE 
Input
  Type: file
  Source: foo.eti
Output
  Name: /dev/null
  Sampling rate: 2.0480 MHz
      Configuration parsed. Starting up version v2.4.1-13-ga2d708d
```

Seems this is in DabMod.cpp#300ff

> mod_settings.useFileOutput

Well, it only works in **Cygwin** when you are using a config file (`odr-dabmod.exe [configfile.ini] > /dev/null`)

```
$ ~/odr-dabmod.exe ~/eti2raw.ini > /dev/null
ODR-DabMod version 2.4.1, compiled at Jan  5 2021, 23:02:14
Compiled with features: SSE
     Input
  Type: file
  Source: /dev/stdin
Output
  Name: /dev/stdout
  Sampling rate: 2.0480 MHz
```

mpbraendli, 2021-01-06:
Huh, maybe a cygwin-specific thing... @LarsHaack are you able to look into that?

LarsHaack, 2021-01-07:
Command line argument parsing has a general issue in the cygwin build.
The call to getopt() does not detect any valid arguments if you start the command line with the input filename.
cygwin's getopt() implementation stops parsing after the first 'non-option' argument. From what I read this seemed to be the standard behavior of getopt() in all initial implementations.
To allow non-option arguments not only at the and of the arguments is an extension that is available in Linux but not in cygwin.

For the current cygwin build it means you have to give the input filename as the last command line argument.

`$ ./odr-dabmod.exe -f /dev/null test.eti`
Works for me.

 ```
    Input
  Type: file
  Source: test.eti
Output
  Name: /dev/null
  Sampling rate: 2.0480 MHz
 Configuration parsed. Starting up version v2.4.1-3-gb977837
ERROR Could not set priority for modulator:134
      Setting up timestamp decoder with 0 offset
      Input file format: raw, length: 30720000, nb frames: 5000
      End of file reached.
      modulator stopped.
```

Unfortunately I have no idea how to get the expected behavior into the cygwin world.

mpbraendli, 2021-01-07:
Thanks for your insight! Maybe the variant we use is a GNU variant, different from POSIX.
We could "fix" this by updating the help screen and documentation, I don't think it's worth investing a lot of time into this.

andimik, 2021-11-28:
I've successfully switched to Windows Subsystem for Linux, so I close this issue as I'm no more interested in Cygwin.

It works like a charm in WSL2. I can create RAW-files from ETI-Files and do not need to reboot to Linux.

closed 2021-11-28 by andimik