Opendigitalradio/ODR-DabMod #47: Build not working

MaxDeepCoder, 2022-02-19:
When i do ./configure

It says this
```
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking target system type... armv7l-unknown-linux-gnueabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
checking for gcc... gcc
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether g++ supports C++11 features with -std=c++11... yes
checking whether C++ compiler accepts -Wduplicated-cond... yes
checking whether C++ compiler accepts -Wduplicated-branches... yes
checking whether C++ compiler accepts -Wlogical-op... yes
checking whether C++ compiler accepts -Wrestrict... yes
checking whether C++ compiler accepts -Wno-pragmas... yes
checking whether C++ compiler accepts -Wdouble-promotion... yes
checking whether C++ compiler accepts "-Wformat=2"... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for FFTW... no
configure: error: FFTW is required

```
```
checking for FFTW... no
configure: error: FFTW is required
```

is the error

Please can you tell me a way to download this dependency

I did 

`sudo apt-get install dev-fftw`

And the problem is still not resloved

MaxDeepCoder, 2022-02-19:
I meant 
sudo apt-get install fftw-dev

andimik, 2022-02-19:
https://packages.ubuntu.com/search?keywords=fftw-dev

You seemed to have installed FFTW2, not FFTW3 (which is required according to `INSTALL.md`).

https://github.com/Opendigitalradio/ODR-DabMod/blob/master/INSTALL.md

Have you tried `sudo apt-get install libfftw3-dev`?

MaxDeepCoder, 2022-02-19:
no

MaxDeepCoder, 2022-02-19:
lemme try that

MaxDeepCoder, 2022-02-19:
Thanks so much !!!!!

MaxDeepCoder, 2022-02-19:
Now I get this

`configure: error: UHD is required`

andimik, 2022-02-19:
I fear you have not read that https://github.com/Opendigitalradio/ODR-DabMod/blob/master/INSTALL.md through ...

As you probably don't use UHD you have to disable it.

MaxDeepCoder, 2022-02-19:
Nice

Thanks for helping me !!!

closed 2022-02-19 by MaxDeepCoder