Opendigitalradio/ODR-DabMod #52: Time is Transmitted without local offset

EndlessEden, 2022-07-22:
dabmod is transmitting the timesync in RTC of the host, on linux this is typically UTC as offsets for local time are done in software. 
However, Hardware DAB Receivers expect Local Time Offset as is normally transmitted by DAB Broadcasters, not UTC, causing receiver to desync.

mpbraendli, 2022-07-22:
ODR-DabMod does not change the time information it receives from ODR-DabMux. The timezone setting of the machine running ODR-DabMod is irrelevant.

ODR-DabMux sets LTO either according to its configuration file, or derives it from the TZ environment variable. See https://github.com/Opendigitalradio/ODR-DabMux/blob/master/doc/example.mux#L108

You may keep your system in UTC, but launch odr-dabmux with the TZ environment variable set to your local timezone and your DAB receivers will show correct time.

closed 2022-07-22 by mpbraendli

EndlessEden, 2022-07-26:
i was unaware it was getting it from dabmux, will look into it. ty