Opendigitalradio/ODR-DabMod #55: Poor u8 support

andimik, 2024-03-03:
The actual version ODR-DabMod version v2.6.0-69 ( g029e9c7 ) has a very bad u8 support. Even with very sensitive AbracaDABra I get only 4.9 dB and poor audio.

[image]

Then I went back to a former installation, ODR-DabMod version v2.6.0-1 (g12f6daf), 12f6dafebf798f4ecc8c80c74221ee9529767e70

Much better, 31 dB

[image]

It's both times the same command.

```
$ cat ./foo.eti | odr-dabmod /dev/stdin -f /tmp/foo.raw -F u8

```

mpbraendli, 2024-03-07:
Interesting observation? Are you able to narrow it down to the commit that changes this behaviour?

andimik, 2024-03-07:
Yes, but only when I find some time. In the meantime I will use the old one.

andimik, 2024-03-07:
But this is the difference (I even did not know it's so easy to compare two commits):

https://github.com/opendigitalradio/ODR-DabMod/compare/029e9c7..12f6daf#diff-4634e877086bf0dad2d6bede99eb945ff1b0dd381662d8dda7b5afb56ee99dcb

check `formatconverter.cpp` and the `for` and `else` , but I have no idea what it means.

mpbraendli, 2024-03-11:
Thanks for notifying me about this, and for your analysis. Can you try latest 'next' ?

andimik, 2024-03-11:
Yes, that 03bf901 fixed the issue :-)

[image]

```
$ cat ./nrk1.eti | odr-dabmod /dev/stdin -f /tmp/nrk1.raw -F u8
ODR-DabMod version v2.6.0-71-g03bf901, compiled at Mar 11 2024, 20:54:28
Compiled with features: zeromq output_soapysdr SSE 
Input
  Type: file
  Source: /dev/stdin
Output
  Name: /tmp/nrk1.raw
  Sampling rate: 2.0480 MHz
```

[image]

closed 2024-03-11 by andimik