Opendigitalradio/ODR-DabMod #57: multiple MUX's on one hackrf one

LaneaLucy, 2024-05-27:
hi, is it possible to send multiple MUX's with one hackrf one? it should be possible if i understand everything right, the question is just can dabmod accept multiple inputs or do i need to run multiple instances of dabmod and add the IQ streams togethere somewehere else? and then the next question is for a example for it.

thx in advance

andimik, 2024-05-27:
Oh, very interesting question. Let's see if there is a solution for this.

closed 2024-05-30 by mpbraendli

LaneaLucy, 2024-05-30:
Why you not want to cover this?
I mean a little example flow graph for people how know how to tinker with it would be great to be included

mpbraendli, 2024-05-30:
I'd love such ideas to be part of the Opendigitalradio project, and that's why there's the discussion forum and the wiki. 

I'm not willing to maintain it myself for lack of time. But please try, tinker, and share your findings!