Opendigitalradio/ODR-DabMod #63: Playing DAB Signals with HackRF One, received signal and parsed SID, but no sound output. Found 'RS errors: uncorrected errors' in welle.io

Locotar, 2025-04-15:
1. System Topology:
HackRF One (Transmitter) → RTL-SDR (Receiver)

2. Software info:
2.1. SoapySDR
 /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.8/libHackRFSupport.so  (0.3.4)
2.2. att@att:~$ odr-dabmod -v
ODR-DabMod version 3.0.0, compiled at Apr  2 2025, 19:09:37
2.3. welle.io
version 2.4
built on 2024-04-05

3. ETI info:
The dab_recording.eti file was recorded using dab2eti, and it can be successfully played back with dablin.

===============================================================
att@att:~/dab/run/0408$ cat d.ini
[input]
transport=file
source=dab_recording.eti
loop=1

[modulator]
rate=2048000
digital_gain=1.0
dabmode=1

[output]
output=soapysdr
device="driver=hackrf,serial=0000000000000000048866dc3746xxxx"

[soapyoutput]
master_clock_rate=10000000
samplerate=4096000

frequency=208064000
txgain=40
antenna=TX/RX

===============================================================
att@att:~/dab/run/0408$ sudo odr-dabmod d.ini
ODR-DabMod version v3.0.0, compiled at Apr  8 2025, 09:26:07
Compiled with features: zeromq output_uhd output_soapysdr NEON
Input
  Type: file
  Source: dab_recording.eti
Output
 SoapySDR
  Device:
  master_clock_rate: 10000000
  Sampling rate: 2.0480 MHz
2025-04-08Z11:15:29       Configuration parsed. Starting up version v3.0.0
2025-04-08Z11:15:29       Running FFTW planning...
2025-04-08Z11:15:33       FFTW planning done.
2025-04-08Z11:15:33       Soapy:Creating the device with:
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

[INFO] Opening HackRF One #0 48866dc3746xxxx...
2025-04-08Z11:15:34       SoapySDR:Actual master clock rate: 0.0000 kHz
2025-04-08Z11:15:34       SoapySDR:Actual TX rate: 2048.0000 ksps.
2025-04-08Z11:15:34       SoapySDR:Actual frequency: 208064.000 kHz.
2025-04-08Z11:15:34       SoapySDR:Actual TX gain: 40.00
2025-04-08Z11:15:34       SoapySDR:Actual TX antenna: TX/RX
2025-04-08Z11:15:34       Setting up timestamp decoder with 0 offset
2025-04-08Z11:15:34       Input file format: raw, length: 22364160, nb frames: 3640
2025-04-08Z11:15:34       Setting up DabModulator...
[... 110 lines cut ...]
                   PrbsGenerator:       8204 us (0.06 %)
                     ConvEncoder:      13728 us (0.10 %)
               PuncturingEncoder:      18823 us (0.13 %)
                 TimeInterleaver:      14927 us (0.10 %)
                FrameMultiplexer:      53713 us (0.37 %)
                BlockPartitioner:      17988 us (0.12 %)
                QpskSymbolMapper:     296003 us (2.05 %)
            FrequencyInterleaver:    1049146 us (7.26 %)
                  PhaseReference:      12993 us (0.09 %)
           DifferentialModulator:    1074191 us (7.44 %)
                      NullSymbol:       5384 us (0.04 %)
                  PhaseReference:       7141 us (0.05 %)
           TII(c:0 p:0 vrnt:new):       6460 us (0.04 %)
               SignalMultiplexer:     426573 us (2.95 %)
                   OfdmGenerator:    3006679 us (20.82 %)
                     GainControl:      30544 us (0.21 %)
           GuardIntervalInserter:     877591 us (6.08 %)
                    OutputMemory:     848895 us (5.88 %)
                           total:   14444464 us (100.00 %)
2025-04-08Z11:15:50       Setting up timestamp decoder with 0 offset
2025-04-08Z11:15:50       Input file format: raw, length: 22364160, nb frames: 3640
2025-04-08Z11:15:50       Setting up DabModulator...
2025-04-08Z11:15:50       DabModulator set up.
......
===============================================================

andimik, 2025-04-15:
Have you tried other eti files?
Why not create an eti file with Qt-DAB or eti-cmdline? 

Could you please test it with (for example) https://github.com/Opendigitalradio/etisnoop/files/3623443/12c_italia_cisco.eti.zip

Locotar, 2025-04-18:
> Have you tried other eti files? Why not create an eti file with Qt-DAB or eti-cmdline?
> 
> Could you please test it with (for example) https://github.com/Opendigitalradio/etisnoop/files/3623443/12c_italia_cisco.eti.zip

Thank you for your suggestion. I have tried using other ETI files—for example, playing them with a DAB signal generator—including the ETI files I recorded myself, and they all worked fine.

###  12c_italia_cisco.eti
```bash
att@att:~/dab/run/0418$ sudo odr-dabmod mod_eti.ini 
ODR-DabMod version v3.0.0, compiled at Apr  8 2025, 09:26:07
Compiled with features: zeromq output_uhd output_soapysdr NEON 
Input
  Type: file
  Source: 12c_italia_cisco.eti
Output
 SoapySDR
  Device: 
  master_clock_rate: 10000000
  Sampling rate: 2.0480 MHz
2025-04-18Z01:52:32       Configuration parsed. Starting up version v3.0.0
2025-04-18Z01:52:32       Running FFTW planning...
2025-04-18Z01:52:36       FFTW planning done.
2025-04-18Z01:52:36       Soapy:Creating the device with: 

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

[INFO] Opening HackRF One #0 48866dc374632c3...
2025-04-18Z01:52:36       SoapySDR:Actual master clock rate: 0.0000 kHz
2025-04-18Z01:52:36       SoapySDR:Actual TX rate: 2048.0000 ksps.
2025-04-18Z01:52:36       SoapySDR:Actual frequency: 227360.000 kHz.
2025-04-18Z01:52:36       SoapySDR:Actual TX gain: 40.00
2025-04-18Z01:52:36       SoapySDR:Actual TX antenna: TX/RX
2025-04-18Z01:52:36       Setting up timestamp decoder with 0 offset
2025-04-18Z01:52:36       Input file format: raw, length: 16128000, nb frames: 2625
2025-04-18Z01:52:36       Setting up DabModulator...
2025-04-18Z01:52:36       DabModulator set up.
2025-04-18Z01:52:48 WARN  ETI FCT discontinuity, expected 202 received 77
2025-04-18Z01:52:48 WARN  Restart modulator.
2025-04-18Z01:52:48       2624 DAB frames, 62.976 seconds encoded
2025-04-18Z01:52:48       Process time:
                    DabModulator:    9654529 us (88.67 %)
                OutputSDR(Soapy):    1233637 us (11.33 %)
                           total:   10888166 us (100.00 %)
2025-04-18Z01:52:48       Process time:
                   PrbsGenerator:     328242 us (3.41 %)
                       FicSource:       5043 us (0.05 %)
                   PrbsGenerator:      12554 us (0.13 %)
                     ConvEncoder:      34776 us (0.36 %)
```
I suspect this might be where the problem lies.
```bash
SoapySDR:Actual master clock rate: 0.0000 kHz
```

[image]

### SoapySDRUtil  shows,  One TCXO is already installed.
```bash
att@att:~/dab/run/0418$ SoapySDRUtil --probe="driver=hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=hackrf
[INFO] Opening HackRF One #0 48866dc3746xxxx...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=HackRF
  hardware=HackRF One
  clock source=external
  part id=a000cb3c007d476e
  serial=0000000000000000048866dc3746xxxx
  version=2023.01.1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO
  Other Settings:
     * Antenna Bias - Antenna port power control.
       [key=bias_tx, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CS8, CS16, CF32, CF64
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Count - Number of buffers per read.
       [key=buffers, units=buffers, default=15, type=int]
  Antennas: TX/RX
  Full gain range: [0, 116] dB
    LNA gain range: [0, 40] dB
    AMP gain range: [0, 14] dB
    VGA gain range: [0, 62] dB
  Full freq range: [0, 7250] MHz
    RF freq range: [0, 7250] MHz
  Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
  Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CS8, CS16, CF32, CF64
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Count - Number of buffers per read.
       [key=buffers, units=buffers, default=15, type=int]
  Antennas: TX/RX
  Full gain range: [0, 61] dB
    VGA gain range: [0, 47] dB
    AMP gain range: [0, 14] dB
  Full freq range: [0, 7250] MHz
    RF freq range: [0, 7250] MHz
  Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
  Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz

att@att:~/dab/run/0418$ 
```

Locotar, 2025-04-18:
I downloaded the Qt-DAB tool. Still using the same receiver — RTL-SDR.
The image below shows the test ETI file on 8B frequency, which was broadcast from our DAB signal generator.
[image]

And this one here is my HackRF One sending out the 12c_italia_cisco.eti file on the 12C frequency.
[image]

andimik, 2025-04-18:
Sorry, if I did not understand you correctly, so you need an errorfree ETI from Riks? Here you are.

This is the source from satellite, so the file is from the broadcaster, not converted from an RTLSDR.

[dab_riks_20250311.eti.zip](https://github.com/user-attachments/files/19811621/dab_riks_20250311.eti.zip)

I have tested it in my lab, but I have no RS errors (tested with PC and with a pocket radio). Could you share your file?

And; I fear you are addressing this issue to the wrong Github project.

Furthermore, could you repeat your file with the following command:

```
cat /tmp/your_file.eti | odr-dabmod  -f /dev/stdout -F s8 -a 0.7 | hackrf_transfer -t /dev/stdin -f 237488000 -x 25 -a 1 -s 2048000 -b 1750000 
```

this if on channel 13E (237.488 MHz)

Locotar, 2025-04-27:
> Sorry, if I did not understand you correctly, so you need an errorfree ETI from Riks? Here you are.
>
> This is the source from satellite, so the file is from the broadcaster, not converted from an [...]

thanks a lot, this commond works for me. not only "12c_italia_cisco.eti" but also all my etis.
```bash
cat /tmp/your_file.eti | odr-dabmod  -f /dev/stdout -F s8 -a 0.7 | hackrf_transfer -t /dev/stdin -f 237488000 -x 25 -a 1 -s 2048000 -b 1750000 
```
but, why? 
ODR-DabMod-master.zip  is [v3.0.0](https://github.com/Opendigitalradio/ODR-DabMod/releases/tag/v3.0.0)

closed 2025-04-30 by Locotar